SharpAI/DeepCamera #8: Survey for development kit supply

solderzzc, 2019-03-11:
Due to the hardware delivery will cost our extra effort, we want to know if this is needed.

Full set development kit will contain:

- Rockchip RK3288 set-up-box with deep camera software 2GB mem, 8GB Flash
- Dahua Camera

Please thumb up if you need it.

chensteven, 2019-03-12:
What is the cost?

solderzzc, 2019-03-12:
@chensteven It cost around  ¥400(RK3288 Box) + ¥120 (Dahua Camera) + ¥100 (extra components) in China. We don't have experience on the handling of shipping/preparation, but would run as non-profit on hardware, estimated $100(hw cost) + $15(shipping) +$20(handling fee).

Paraesthetic, 2019-03-13:
Sounds interesting

samsomlan, 2019-03-19:
how to contact you?  I have a lot  interesting about your project。

solderzzc, 2019-03-19:
@samsomlan please join our slack ;)

https://sharpai-invite-automation.herokuapp.com/

closed 2020-09-17 by solderzzc