SharpAI/DeepCamera #11: My side need to support the s3 storage or other

yufeng2015, 2019-03-13:
**hello!**

> My side need to support the s3 storage or other, but need to know the current logic of uploading files, so that together

1. Only the following file?
> https://github.com/SharpAI/DeepCamera/blob/master/src/detector/upload.js

2. The following two is the key method?
 
> [image]

solderzzc, 2019-03-13:
The upload interface is defined here:
https://github.com/SharpAI/DeepCamera/blob/master/src/detector/upload.js#L127

you can add a new file to implement it, after then, import in the up layer module with ENV control or setting file.

yufeng2015, 2019-03-13:
‘’putFileAliyun‘’，‘ getAccessUrl’，Who is calling them?

solderzzc, 2019-03-13:
please reopen if still have question.

closed 2019-03-13 by solderzzc