SharpAI/DeepCamera #12: Better site UI? (ymmv)

macginitie, 2019-03-14:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when line breaks are funky & tenses are incosistent.

**Describe the solution you'd like**
Change the text in the 2nd box in lifecycle.png: from PreProc [\n] essing to Pre- [\n] Process

... because it's not Labeling, Training, Model Deploying

**Additional context**
[image]

macginitie, 2019-03-14:
incoNsistent :-p

solderzzc, 2019-03-14:
Oh, @macginitie  that's great suggestion :)

solderzzc, 2019-03-14:
I was thinking about this ugly thing in 1sec, but no better ides :)

macginitie, 2019-03-14:
I'll just say, I like the clean look of all boxes being the same size :)

solderzzc, 2019-03-14:
@macginitie Thanks a lot, just updated :) Reopen if still has issue lol

solderzzc, 2019-03-14:
hmm, the CDN cache still loading the previous image.

solderzzc, 2019-03-14:
Yep, done for now. please check @macginitie

commit 7d6cb865 (2019-03-14): To fix #12
commit 7fe6b40e (2019-03-14): To fix #12
closed 2019-03-14 by solderzzc