SharpAI/DeepCamera #15: pkg update failed

lambda-zhang, 2019-03-14:
```
$ pkg update
Hit:1 https://termux.net stable InRelease
Hit:2 https://its-pointless.github.io/files termux InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
31 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libicu
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  termux-auth
The following packages will be upgraded:
  apt binutils busybox c-ares cmake command-not-found dpkg gdbm git gnupg gpgv graphicsmagick libgcrypt libmosquitto libnghttp2 libsqlite libtiff libuv make mosquitto ncurses ncurses-ui-libs nodejs-lts
  openssh openssl python2-numpy python2-scipy redis setup-scripts termux-tools util-linux
31 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 31.7 MB of archives.
After this operation, 2826 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://termux.net stable/main aarch64 busybox aarch64 1.29.3-1
  404  Not Found
Err:2 https://termux.net stable/main aarch64 dpkg aarch64 1.19.2
  404  Not Found
Get:3 https://termux.net stable/main aarch64 libgcrypt aarch64 1.8.4 [272 kB]
Err:4 https://its-pointless.github.io/files termux/extras aarch64 python2-numpy aarch64 1.15.4
  404  Not Found
Err:5 https://its-pointless.github.io/files termux/extras aarch64 python2-scipy aarch64 1.2.0rc1
  404  Not Found
Err:6 https://termux.net stable/main aarch64 gpgv aarch64 2.2.11
  404  Not Found
Get:7 https://its-pointless.github.io/files termux/extras all setup-scripts all 2.6.2 [2632 B]
Err:8 https://termux.net stable/main aarch64 apt aarch64 1.4.8-7
  404  Not Found
Err:9 https://termux.net stable/main all termux-tools all 0.58
  404  Not Found
Get:10 https://termux.net stable/main aarch64 command-not-found aarch64 1.36 [23.7 kB]
Err:11 https://termux.net stable/main aarch64 libnghttp2 aarch64 1.35.0
  404  Not Found
Err:12 https://termux.net stable/main aarch64 ncurses aarch64 6.1.20181117
  404  Not Found
Err:13 https://termux.net stable/main aarch64 openssl aarch64 1.1.1a
  404  Not Found
Err:14 https://termux.net stable/main aarch64 binutils aarch64 2.31.1-3
  404  Not Found
Get:15 https://termux.net stable/main aarch64 c-ares aarch64 1.15.0 [23.1 kB]
Err:16 https://termux.net stable/main aarch64 libuv aarch64 1.24.0
  404  Not Found
Get:17 https://termux.net stable/main aarch64 make aarch64 4.2.1-2 [77.8 kB]
Err:18 https://termux.net stable/main aarch64 cmake aarch64 3.13.1
  404  Not Found
Get:19 https://termux.net stable/main aarch64 gdbm aarch64 1.18.1 [52.0 kB]
Err:20 https://termux.net stable/main aarch64 git aarch64 2.19.2                                                                                                                                            
  404  Not Found
Err:21 https://termux.net stable/main aarch64 libsqlite aarch64 3.25.3                                                                                                                                      
  404  Not Found
Err:22 https://termux.net stable/main aarch64 gnupg aarch64 2.2.11                                                                                                                                          
  404  Not Found
Err:23 https://termux.net stable/main aarch64 libtiff aarch64 4.0.10-1                                                                                                                                      
  404  Not Found
Err:24 https://termux.net stable/main aarch64 graphicsmagick aarch64 1.3.31                                                                                                                                 
  404  Not Found
Err:25 https://termux.net stable/main aarch64 libmosquitto aarch64 1.5.4                                                                                                                                    
  404  Not Found
Err:26 https://termux.net stable/main aarch64 mosquitto aarch64 1.5.4                                                                                                                                       
  404  Not Found
Err:27 https://termux.net stable/main aarch64 ncurses-ui-libs aarch64 6.1.20181117                                                                                                                          
  404  Not Found
Err:28 https://termux.net stable/main aarch64 nodejs-lts aarch64 8.13.0                                                                                                                                     
  404  Not Found
Get:29 https://termux.net stable/main aarch64 termux-auth aarch64 1.1 [3748 B]                                                                                                                              
Err:30 https://termux.net stable/main aarch64 openssh aarch64 7.9p1-1                                                                                                                                       
  404  Not Found
Err:31 https://termux.net stable/main aarch64 redis aarch64 5.0.2                                                                                                                                           
  404  Not Found
Err:32 https://termux.net stable/main aarch64 util-linux aarch64 2.33-1                                                                                                                                     
  404  Not Found
Fetched 455 kB in 10s (43.8 kB/s)                                                                                                                                                                           
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/busybox_1.29.3-1_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/dpkg_1.19.2_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/gpgv_2.2.11_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/apt_1.4.8-7_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-all/termux-tools_0.58_all.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/libnghttp2_1.35.0_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/ncurses_6.1.20181117_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/openssl_1.1.1a_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/binutils_2.31.1-3_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/libuv_1.24.0_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/cmake_3.13.1_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/git_2.19.2_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/libsqlite_3.25.3_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/gnupg_2.2.11_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/libtiff_4.0.10-1_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/graphicsmagick_1.3.31_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/libmosquitto_1.5.4_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/mosquitto_1.5.4_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/ncurses-ui-libs_6.1.20181117_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/nodejs-lts_8.13.0_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/openssh_7.9p1-1_aarch64.deb  404  Not Found
E: Failed to fetch https://its-pointless.github.io/files/dists/termux/extras/binary-aarch64/python2-numpy_1.15.4_aarch64.deb  404  Not Found
E: Failed to fetch https://its-pointless.github.io/files/dists/termux/extras/binary-aarch64/python2-scipy_1.2.0rc1_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/redis_5.0.2_aarch64.deb  404  Not Found
E: Failed to fetch https://termux.net/dists/stable/main/binary-aarch64/util-linux_2.33-1_aarch64.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

```

lambda-zhang, 2019-03-14:
solved by:
```
$ rm -rf /data/data/com.termux/files/usr/var/lib/apt/*
$ pkg update
```

solderzzc, 2019-03-15:
Fixed

closed 2019-03-15 by solderzzc