SharpAI/DeepCamera #17: Run_On_Android_aarch64 js error

yufeng2015, 2019-03-14:
https://github.com/SharpAI/DeepCamera/blob/master/docs/Run_On_Android_aarch64.md

The last，run the command
```
cd /data/data/com.termux/files/home/runtime
termux-wake-lock
setsid bash ~/runtime/start_aarch64.sh
```

**Error is as follows（Because the camera didn't open?）**

```
  return binding.read(fd, buffer, offset, length, position);
                 ^

Error: EINVAL: invalid argument, read
    at Object.fs.readSync (fs.js:675:18)
    at tryReadSync (fs.js:540:20)
    at Object.fs.readFileSync (fs.js:575:19)
    at I.os (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7234)
    at Timeout._onTimeout (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7447)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
[2019-03-14 12:23:26,762: WARNING/ForkPoolWorker-1] ['workaicmd', 'autogroup_dataset', 'sync_autogroup_dataset', 'autogroup', '/msg/d/b65772ac1e4f']
[2019-03-14 12:23:26,763: WARNING/ForkPoolWorker-1] ['/device/e287250eca0f2eca592500f1']
[2019-03-14 12:23:26,764: WARNING/ForkPoolWorker-1] ['/msg/g/e287250eca0f2eca592500f1']
exit index.js
[2019-03-14 12:23:26,844: WARNING/ForkPoolWorker-1] mqtt connected, rc: 0
[2019-03-14 12:23:26,961: WARNING/ForkPoolWorker-1] workaicmd
[2019-03-14 12:23:26,964: WARNING/ForkPoolWorker-1] autogroup_dataset
[2019-03-14 12:23:26,967: WARNING/ForkPoolWorker-1] sync_autogroup_dataset
[2019-03-14 12:23:26,972: WARNING/ForkPoolWorker-1] autogroup
[2019-03-14 12:23:26,975: WARNING/ForkPoolWorker-1] /msg/d/b65772ac1e4f
[2019-03-14 12:23:26,978: WARNING/ForkPoolWorker-1] /device/e287250eca0f2eca592500f1
[2019-03-14 12:23:26,982: WARNING/ForkPoolWorker-1] /msg/g/e287250eca0f2eca592500f1
entering index.js
./start_monitor.sh: line 19: pushd: monitor: No such file or directory
new connection to meteor server
4314ef9cb9fab3766e9918fe7a55e7b905e265d99a88026fa110841182426e4d
login ok:qCA5imBb-PeyQJwqRAE6_vBJ_dFiSCeK0wUKDUiyK-G
commands complete:
undefined
[ADDED] to devices:  -eabad615d9a3f665de6b9bb6
{ '-eabad615d9a3f665de6b9bb6': 
   { _id: '-eabad615d9a3f665de6b9bb6',
     uuid: 'b65772ac1e4f',
     name: 'b65772ac1e4f开发板',
     in_out: 'inout',
     groupId: 'e287250eca0f2eca592500f1',
     createAt: 2019-03-04T13:26:30.047Z,
     online: true,
     camera_run: false,
     autoUpdate: false } }
yes, my group id is [e287250eca0f2eca592500f1] for now
watchtower disabled
commands complete:
{ '-eabad615d9a3f665de6b9bb6': 
   { _id: '-eabad615d9a3f665de6b9bb6',
     uuid: 'b65772ac1e4f',
     name: 'b65772ac1e4f开发板',
     in_out: 'inout',
     groupId: 'e287250eca0f2eca592500f1',
     createAt: 2019-03-04T13:26:30.047Z,
     online: true,
     camera_run: false,
     autoUpdate: false } }
The file was saved!
fs.js:675
  return binding.read(fd, buffer, offset, length, position);
[... 509 lines cut ...]
$ 
$ 
$ 
$ 
$ 
$ 
$ 
$ 
$ 
$ fs.js:675
  return binding.read(fd, buffer, offset, length, position);
                 ^

Error: EINVAL: invalid argument, read
    at Object.fs.readSync (fs.js:675:18)
    at tryReadSync (fs.js:540:20)
    at Object.fs.readFileSync (fs.js:575:19)
    at I.os (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7234)
    at Timeout._onTimeout (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7447)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
exit index.js

```

solderzzc, 2019-03-14:
You can run monitor/src directly to debug the root cause of this error message.

solderzzc, 2019-03-14:
I have the same issue here... Need to dig into it.

lambda-zhang, 2019-03-15:
```
$ cat /sys/class/thermal/thermal_zone*/temp
$ cat /sys/class/thermal/thermal_zone*/type
```

If got this meams  kernel bug of thermal
```
cat: /sys/class/thermal/thermal_zone0/temp: Invalid argument
```

Have try with #20

yufeng2015, 2019-03-15:
Hardware problem?Didn't quite understand

solderzzc, 2019-03-15:
It's fixed by this commit:
https://github.com/SharpAI/DeepCamera/pull/20

Please pull latest code, let me close this issue.

Reopen if still have the same problem.

closed 2019-03-15 by solderzzc