SharpAI/DeepCamera #21: rk3399 ubuntu 16.04 build

luqinlive, 2019-03-15:
how can i build this project in rk3399 ubuntu 16.04 , to detected faces and upload to ftpserver , is it enable hardware accelerated to decode 1080p rtsp video,thank you!

cici-tan, 2019-03-15:
Check out our docs folder: https://github.com/SharpAI/DeepCamera/blob/master/docs/Run_On_Android_aarch64.md

luqinlive, 2019-03-15:
> 
> 
> Check out our docs folder: https://github.com/SharpAI/DeepCamera/blob/master/docs/Run_On_Android_aarch64.md

Thank you very much for your quick reply, I knew is was Android doc, can it be installed on Ubuntu Linux?

luqinlive, 2019-03-15:
i build face detector   and run  start_detector.sh  ,got this error 

init nedb global
init nedb global for gif
init nedb global for timeline
upload listerer is ready
wait detect queue init
uncaughtException { Error: listen EADDRINUSE 0.0.0.0:3000
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at Server.setupListenHandle [as _listen2] (net.js:1367:14)
    at listenInCluster (net.js:1408:12)
    at doListen (net.js:1517:7)
    at _combinedTickCallback (internal/process/next_tick.js:141:11)
    at process._tickDomainCallback (internal/process/next_tick.js:218:9)
    at Function.Module.runMain (module.js:695:11)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 3000 }
sharpai bucket does not exist
{ Error: connect ECONNREFUSED 192.168.10.124:9000
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1198:14)
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.10.124',
  port: 9000 }
Error creating bucket. { Error: connect ECONNREFUSED 192.168.10.124:9000
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1198:14)
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.10.124',
  port: 9000 }
uncaughtException { Error: connect ECONNREFUSED 192.168.10.124:9000
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1198:14)
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.10.124',
  port: 9000 }
uncaughtException { Error: socket hang up
    at createHangUpError (_http_client.js:331:15)
    at Socket.socketCloseListener (_http_client.js:363:23)
    at emitOne (events.js:121:20)
    at Socket.emit (events.js:211:7)
    at TCP._handle.close [as _onclose] (net.js:567:12) code: 'ECONNRESET' }
select database 15 to avoid conflict
The connection to celery cluster is connected
The connection to celery broker is connected

solderzzc, 2019-03-15:
Lol, @cici-tan 

Hi, @luqinlive 
Please use this docker file on ubuntu rk3399:
https://github.com/SharpAI/DeepCamera/blob/master/docker/docker-compose-prebuilt.yml

Just simply run:
https://github.com/SharpAI/DeepCamera/blob/master/docker/run-deepeye-prebuilt.sh

```
cd docker
./run-deepeye-prebuilt.sh start
```

If you want to build the docker file locally, you will need to check this repo:
https://github.com/SharpAI/ImageBuilder-DeepLearning

https://github.com/SharpAI/ImageBuilder-DeepLearning/blob/android_porting/Docker-DeepEye/workai/build-deepeye.sh

kaishijeng, 2019-03-17:
How do you install docker on rk3399?
I have firefly rk3399 board with ubuntu installed.
However, it has the following error when I install docker.io:

Job for docker.service failed because the control process exited with error code. See "systemctl status docker.service" and "journalctl -xe" for details.

My ubuntu is:

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.2 LTS
Release:        16.04
Codename:       xenial

Kernel:

Linux firefly 4.4.52 #19 SMP Tue May 2 11:36:30 HKT 2017 aarch64 aarch64 aarch64 GNU/Linux

solderzzc, 2019-03-17:
your kernel might have no docker support, you can either  compile kernel with docker support or switch to android on rk3399
https://github.com/SharpAI/DeepCamera/blob/master/docs/Run_On_Android_aarch64.md

kaishijeng, 2019-03-17:
@solderzzc 

Thanks for the info. However, I prefer to use ubuntu since I already have other libraries installed.
Any info/link which I can follow to build new kernel which is docker compatible

Thanks

solderzzc, 2019-03-17:
@kaishijeng

You may want to check this script:

https://raw.githubusercontent.com/docker/docker/master/contrib/check-config.sh

kaishijeng, 2019-03-18:
There are a lot of missing features in kernel when I ran check_config.sh
I tried to follow android instructions on H96Max-RK3399. Able to run termux and ssh into it.
Download usr_aarch64_dev_1204_2018.tgz and untar into machine.
However, I got errors when run
LD_LIBRARY_PATH=/system/lib64:$LD_LIBRARY_PATH:$PREFIX/lib64:/system/vendor/lib64/egl:/system/vendor/lib64 python2

import tvm

OSError: dlopen failed: library "libOpenCL.so" not found

import mxnet

OSError: dlopen failed: library "libGLES_mali.so" not found

Do you know where these 2 *.so files are located?

solderzzc, 2019-03-18:
it should be placed here: /system/vendor/lib64/egl

how about your Android？

kaishijeng, 2019-03-18:
ls -lt /system/vendor/lib64/
total 40
drwxr-xr-x 2 root shell  4096 Jun 11  2018 mediadrm
-rw-r--r-- 1 root root  35264 Jun 11  2018 libbt-vendor.so

My system is:
H96 MAX RK3399 Hexa Core 2.0Ghz 4K TV BOX Android 7.1 4GB/32GB Type-C USB3.0 AC WIFI 1000M LAN Bluetooth

solderzzc, 2019-03-18:
We have all in one images to run on H96 Max RK3399, the system is Linux and docker started automatically.

https://github.com/SharpAI/facebox_sdk/tree/master/firmware

solderzzc, 2019-03-18:
libGLES_mali.so is placed in  /system/vendor/lib64/egl, verified in 4 Android(aarch64):
1. Samsung 7420
2. MTK6797
3. Kirin 970
4. RK3399 Android Rockpro64 7.1.2

kaishijeng, 2019-03-18:
I need to buy a type-A to type-A cable to use your image.
One thing is not clear when to "Push reset button".
DO I need to push reset button and continue to hold before power plugin?
Or Plugin power first and then push reset button and hold it until upgrade finishes?

solderzzc, 2019-03-18:
It mean push reset button and continue to hold before power plugin.

kaishijeng, 2019-03-18:
I got this error when I ran 
sudo ./upgrade_tool ./qdh_upd96max_201811292100_7d8321e_realtime.img

No found any rockusb device,please plug device in!

Should I continue to hold reset button after power on?

solderzzc, 2019-03-18:
Hmm, you should push till the system saw USB device.

solderzzc, 2019-03-18:
Are you using virtual machine? if so make sure the box is connected to virtual system.

kaishijeng, 2019-03-21:
I got a new type A to type A USB cable and my linux PC can detect H96box now.
However, I got an error when I run sudo ./upgrade_tool uf imagefile.
It burns to flash  for a while and then complains uploaded filed. Try 2 times and the result is the same.
Do you know any win utility which I can use to burn your ubuntu image?

Thanks,

kaishijeng, 2019-03-22:
@solderzzc 

This image, qdh_update_h96max_201811292100_7d8321e_realtime.img seems too big (5GB) for flash burner to handle. I use upgrade_tool on linux and it fails to download to H96. 
On Win PC, I use RokcChip Batch tool, it fails to read this image. Other images can be read by Batch Tool.
So the issue may be h96max_201811292100_7d8321e_realtime.img is too big for tools to handle

solderzzc, 2019-03-22:
@kaishijeng yes, you must use the tool `upgrade_tool` on ubuntu to download, there might be one bug you need to workaround: when the log on console saying: `reset/rebooting`, you need manually reset the hardware and connect back immediately. It works even on my virtual box ubuntu.

kaishijeng, 2019-03-22:
My issue is different. See below for the log of upgrade_tool. One thing to note is image total reported by upgrade_tool which is 1147701K, much smaller than th eimage itself

Loading firmware...
Support Type:RK330C     FW Ver:6.0.01   FW Time:2018-11-29 20:59:19
Loader ver:1.05 Loader Time:2017-05-16 15:38:17
Download Image Total(1147701K),Current(150389K)
Download Firmware Fail

solderzzc, 2019-03-22:
Might cause by two reasons:
1. the cable is not stable.
2. the heat of H96 MAX, you may need cool it down.

solderzzc, 2019-03-22:
Another information, it always works on my MacBook with Ubuntu 18.04.2 on VirtualBox.

kaishijeng, 2019-03-22:
Why does upgrade_tool reports image size: 1147701K?
Is this normal?
I use the same cable and do android flash burning, it works OK every time.
So I don't think the cable is an issue. Not sure about heat of the box

solderzzc, 2019-03-22:
you need contact Rockchip.
https://github.com/rockchip-linux

kaishijeng, 2019-03-23:
I am able to bring up debian on H96 Max via SD card.
However, docker is still not running.
Can I install and use your SW without docker on my H96Max?

Thanks,

kaishijeng, 2019-03-23:
@solderzzc 

Can you share kernel binary and kernel modules in the qdh_update_h96max_201811292100_7d8321e_realtime.img?
Or tell me how to extract them from qdh image?

Thanks

solderzzc, 2019-03-24:
@kaishijeng I moved the rk3399-kernel into SharpAI repo:
https://github.com/SharpAI/rk3399-kernel

@lambda-zhang thanks for your contribution, your work is currently under SharpAI repo.

kaishijeng, 2019-03-24:
@lambda-zhang 

Is this supposed to build on a x64 or aarch64 system?
I did .'make.sh sz3390' on rk3399 debian system and got the following error:

Makefile:708: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
scripts/kconfig/conf  --silentoldconfig Kconfig
Makefile:708: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CC      scripts/mod/empty.o
  CC      scripts/mod/devicetable-offsets.s
  CHK     include/generated/timeconst.h
  CC      kernel/bounds.s
./../prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-gcc -Wp,-MD,scripts/mod/.empty.o.d -nostdinc -isystem ./../prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-gcc -print-file-name=include [Errno 8] Exec format error -I./arch/arm64/include -Iarch/arm64/include/generated/uapi -Iarch/arm64/include/generated -Iinclude -I./arch/arm64/include/uapi -Iarch/arm64/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -mlittle-endian -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -mgeneral-regs-only -fno-pic -fno-asynchronous-unwind-tables -Os -DCC_HAVE_ASM_GOTO -fstack-protector-strong -fno-omit-frame-pointer -fno-optimize-sibling-calls -g -DKBUILD_STR(s)=#s -DKBUILD_BASENAME=KBUILD_STR(empty) -DKBUILD_MODNAME=KBUILD_STR(empty) -c -o scripts/mod/.tmp_empty.o scripts/mod/empty.c [Errno 8] Exec format error
./../prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-gcc -Wp,-MD,scripts/mod/.devicetable-offsets.s.d -nostdinc -isystem ./../prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-gcc -print-file-name=include [Errno 8] Exec format error -I./arch/arm64/include -Iarch/arm64/include/generated/uapi -Iarch/arm64/include/generated -Iinclude -I./arch/arm64/include/uapi -Iarch/arm64/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -mlittle-endian -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -mgeneral-regs-only -fno-pic -fno-asynchronous-unwind-tables -Os -DCC_HAVE_ASM_GOTO -fstack-protector-strong -fno-omit-frame-pointer -fno-optimize-sibling-calls -g -DKBUILD_STR(s)=#s -DKBUILD_BASENAME=KBUILD_STR(devicetable_offsets) -DKBUILD_MODNAME=KBUILD_STR(devicetable_offsets) -fverbose-asm -S -o scripts/mod/devicetable-offsets.s scripts/mod/devicetable-offsets.c [Errno 8] Exec format error
scripts/Makefile.build:266: recipe for target 'scripts/mod/empty.o' failed
make[2]: *** [scripts/mod/empty.o] Error 8
make[2]: *** Waiting for unfinished jobs....
scripts/Makefile.build:153: recipe for target 'scripts/mod/devicetable-offsets.s' failed
make[2]: *** [scripts/mod/devicetable-offsets.s] Error 8
scripts/Makefile.build:411: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
Makefile:573: recipe for target 'scripts' failed
make: *** [scripts] Error 2
make: *** Waiting for unfinished jobs....

kaishijeng, 2019-03-26:
Seems nothing working on my H96Max box. Is it possible to convert your image to a SD image so it can be booted from SD card? It is much easier than download to an emmc flash memory.

Thanks,

closed 2019-04-03 by solderzzc