SharpAI/DeepCamera #23: New Deepvision model addition

Ritesh1991, 2019-03-17:
what should we do if we want to add new model into your platform?

solderzzc, 2019-03-17:
@Ritesh1991  

We need better document for your question, before done, let me go through the procedure quickly:

1. all image will be passed into `detector` (nodejs for state and web/io operation)
https://github.com/SharpAI/DeepCamera/blob/master/src/detector/index.js#L512

2. we use celery to build worker for easy maintain and future distributed system

For example, we use `embedding` work to get feature of preprocessed images through model inference.

- https://github.com/SharpAI/DeepCamera/blob/master/src/detector/deepeye.js#L190
- https://github.com/SharpAI/DeepCamera/blob/master/src/embedding/upload_api-v2.py#L1661

3. then you might need classifier, the work flow is the same as `embedding`.

Currently, we have three task/worker:
1. face detection(Preprocess)
2. extract face embedding (model inference)
3. classifier

solderzzc, 2019-03-17:
Another question is `how to run model on ARM GPU`, we leveraged TVM, for the face recognition, we documented [here](https://github.com/SharpAI/fast_ai/tree/master/fast_facenet#how-to-run-arm-mali-benchmark)

After converted, integrated as following:
- https://github.com/SharpAI/DeepCamera/blob/master/src/embedding/FaceProcessing.py#L111
- https://github.com/SharpAI/DeepCamera/blob/master/src/embedding/FaceProcessing.py#L198

The model train/test code is from insight face (mxnet):
https://github.com/deepinsight/insightface/blob/master/deploy/face_embedding.py

Compile mxnet into tvm document:
https://docs.tvm.ai/tutorials/nnvm/from_mxnet.html

That's not a lot of works, lol. Just took months for us to figure out the whole procedure which works best.

We also need better architecture to easily integrate new model :)

Ritesh1991, 2019-03-17:
thanks for quick reply. You people did great work.I will try to use with different model. lets see what happen. and do you have any video of how to use app ?

solderzzc, 2019-03-17:
thanks, @cici-tan is compiling the video. Will release soon.

cici-tan, 2019-03-18:
Hey @Ritesh1991 you can check this first https://github.com/SharpAI/mobile_app_server/blob/android_porting/README.md
our video for app is in process

closed 2019-04-03 by solderzzc