SharpAI/DeepCamera #28: run on 3288 ，{ Error: ENOENT: no such file or directory, open ''

yufeng2015, 2019-03-22:
error:

```
[device] tid: 1553216425867 person num: 1 face num: 1 cost: 1405
[2019-03-22 01:00:30,728: INFO/ForkPoolWorker-1] Task upload_api-v2.detect[e7326856-9353-42fd-9157-d4a2a171124c] succeeded in 1.377794542000629s: '{"detected": true, "ts": 1553216429325, "totalPeople": 1, "cropped": [{"path": "/mnt/internal_sd/Download/frame_22032019_090029_131_0.png", "style": "right_side", "blury": 1869.5949723385252, "ts": 1553216429325, "trackerid": 1553216425867, "totalPeople": 1, "cameraId": "device", "width": 179, "height": 180}], "totalmtcnn": 1}'
[2019-03-22 09:00:30,744: INFO/MainProcess] Received task: upload_api-v2.extract_v2[d8d78e01-5bcb-4460-951a-2876329ca15a]   expires:[2019-03-22 01:01:00.000729+00:00]
******cluster_client.call, result******
[2019-03-22 09:00:30,776: INFO/ForkPoolWorker-1] Task upload_api-v2.extract_v2[d8d78e01-5bcb-4460-951a-2876329ca15a] succeeded in 0.0235482919961s: '{"embedding_path": ""}'
uncaughtException { Error: ENOENT: no such file or directory, open ''
    at Object.fs.openSync (fs.js:646:18)
    at Object.fs.writeFileSync (fs.js:1299:33)
    at Result.<anonymous> (/data/data/com.termux/files/home/runtime/main.bin.js:1:12188)
    at emitOne (events.js:116:13)
    at Result.emit (events.js:211:7)
    at RedisClient.<anonymous> (/data/data/com.termux/files/home/runtime/node_modules/node-celery/celery.js:180:21)
    at emitThree (events.js:136:13)
    at RedisClient.emit (events.js:217:7)
    at return_pub_sub (/data/data/com.termux/files/home/runtime/node_modules/redis/index.js:794:18)
    at RedisClient.return_reply (/data/data/com.termux/files/home/runtime/node_modules/redis/index.js:833:9) errno: -2, code: 'ENOENT', syscall: 'open', path: '' }
/mnt/internal_sd/Download/frame_22032019_090030_831.jpg
[2019-0
```

solderzzc, 2019-03-22:
it seems a bug which was fixed 1-2 days before, are you running the latest code?

yufeng2015, 2019-03-22:
The same code, the exception is 3399, after the box to the monitoring group, there is no this exception, currently 3399 only the following exception, but does not affect work

Today I have more test

```
>>>>gifQueue.process,src/detector/index.js
entering index.js
~/runtime/monitor ~/runtime
DDP connection error!
~/runtime
exit index.js

```

solderzzc, 2019-03-22:
Is that possible to build new code for a test?
I think this commit fixed it:
https://github.com/SharpAI/DeepCamera/commit/4580203badfba93675aba7ca9273f126b60b1280#diff-cf1b5b800f84daa697fd456f1f41a096R508

Correct me if I'm wrong.

solderzzc, 2019-03-22:
Latest code will no longer depending on embedding_path returning from celery worker which was lazy temp solution.

yufeng2015, 2019-03-22:
Ok, I will test again

closed 2019-03-22 by yufeng2015