SharpAI/DeepCamera #30: The latest edition of the code ，run on 3399

yufeng2015, 2019-03-22:
read TEMP failed
@solderzzc 

```

[2019-03-22 14:56:21,023: INFO/ForkPoolWorker-1] Task upload_api-v2.detect[a5b20244-8782-4d87-9e66-b7b5a9e7ef6e] succeeded in 0.140305097s: '{"totalPeople": 0, "detected": false, "totalmtcnn": null, "ts": 1553237780843, "cropped": []}'
/storage/emulated/0/Download/frame_22032019_145620_932.jpg
[2019-03-22 14:56:21,076: INFO/MainProcess] Received task: upload_api-v2.detect[07f305a0-4bcd-4a7e-b9e5-6fede78ba017]   expires:[2019-03-22 06:56:51.000062+00:00]
[2019-03-22 14:56:21,253: INFO/ForkPoolWorker-1] Task upload_api-v2.detect[07f305a0-4bcd-4a7e-b9e5-6fede78ba017] succeeded in 0.141344013999s: '{"totalPeople": 0, "detected": false, "totalmtcnn": null, "ts": 1553237781061, "cropped": []}'
[device] tid: 1553237781050 person num: 0 face num: 0 cost: 206
read TEMP failed  { Error: EINVAL: invalid argument, read
    at Object.fs.readSync (fs.js:675:18)
    at tryReadSync (fs.js:540:20)
    at Object.fs.readFileSync (fs.js:575:19)
    at A.os (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7279)
    at Timeout._onTimeout (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7536)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) errno: -22, code: 'EINVAL', syscall: 'read' }
read TEMP failed  { Error: EINVAL: invalid argument, read
    at Object.fs.readSync (fs.js:675:18)
    at tryReadSync (fs.js:540:20)
    at Object.fs.readFileSync (fs.js:575:19)
    at A.os (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7279)
    at Timeout._onTimeout (/data/data/com.termux/files/home/runtime/monitor/main.bin.js:1:7536)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) errno: -22, code: 'EINVAL', syscall: 'read' }
/storage/emulated/0/Download/frame_22032019_145650_415.jpg

```

solderzzc, 2019-03-22:
@yufeng2015 please try this script instead of build it: 
https://github.com/SharpAI/DeepCamera/blob/master/start_service.sh

solderzzc, 2019-03-29:
fixed.

closed 2019-03-29 by solderzzc