SharpAI/DeepCamera #31: The latest code，3288 packing， always  ‘No space left on device’

yufeng2015, 2019-03-22:
```
OSError: [Errno 28] No space left on device
cp: cannot stat '../src/embedding/judgeutil.so_termux_arch-linux_arm': No such file or directory
cp: cannot stat 'dist/classifier/*': No such file or directory
mkdir: cannot create directory '.termux/boot/': No space left on device
mkdir: cannot create directory '.minio': No space left on device
cp: cannot create regular file '.minio/': Not a directory
cp: cannot create regular file '.termux/boot/': Not a directory
chmod: cannot access '.termux/boot/termux_boot_sharpai_arm.sh': No such file or directory
creating sharpai-app.tgz ...
tar: .minio: No such file or directory
gzip: write error: No space left on device
tar: write error: Broken pipe
-rw------- 1 u0_a69 u0_a69 0 Mar 22 15:18 sharpai-app.tgz
$ 
$ 
```

solderzzc, 2019-03-22:
how about `df` or `df -h`, it always caused by `no space left on device`...

solderzzc, 2019-03-29:
close due to no activity.

closed 2019-03-29 by solderzzc