SharpAI/DeepCamera #33: 3399 cluster, the master and the worker how to configure？

yufeng2015, 2019-03-26:
3399 cluster, the master and the worker how to configure

I here the current configuration as follows, but no message processing

master(192.168.132.75):
```
$ cat runtime/embedding_aarch64.sh
#!/usr/bin/env bash
export RUNTIME_BASEDIR=`pwd`
export LD_LIBRARY_PATH=/system/lib64:$LD_LIBRARY_PATH:$PREFIX/lib64:/system/vendor/lib64/egl:/system/vendor/lib64
export DATA_RUNTIME_FOLDER=../model
export DEVICE_UUID_FILEPATH=/data/data/com.termux/files/home/.ro_serialno
export DEVICE_GROUP_ID_FILEPATH=/data/data/com.termux/files/home/.groupid.txt
export RUNTIME_BASEDIR=/data/data/com.termux/files/home/runtime
export CLUSTER_REDIS_ADDRESS=192.168.132.35
export CLUSTER_REDIS_PORT=6379
export CLUSTER_CONCURRENCY=2
cd bin
while [ 1 ]
do
    WORKER_TYPE=embedding REDIS_ADDRESS=localhost WORKER_BROKER=redis://localhost/0 ./embedding worker --loglevel INFO -E -n embedding -c 1 -Q embedding
    sleep 20
done
```

worker(192.168.132.132):
```
$ cat runtime/embedding_aarch64.sh
#!/usr/bin/env bash
export RUNTIME_BASEDIR=`pwd`
export LD_LIBRARY_PATH=/system/lib64:$LD_LIBRARY_PATH:$PREFIX/lib64:/system/vendor/lib64/egl:/system/vendor/lib64
export DATA_RUNTIME_FOLDER=../model
export DEVICE_UUID_FILEPATH=/data/data/com.termux/files/home/.ro_serialno
export DEVICE_GROUP_ID_FILEPATH=/data/data/com.termux/files/home/.groupid.txt
export RUNTIME_BASEDIR=/data/data/com.termux/files/home/runtime
export CLUSTER_REDIS_ADDRESS=192.168.132.35
export CLUSTER_REDIS_PORT=6379
export CLUSTER_CONCURRENCY=2
export CLUSTER_WORKERONLY=True
cd bin
while [ 1 ]
do
    WORKER_TYPE=embedding REDIS_ADDRESS=localhost WORKER_BROKER=redis://localhost/0 ./embedding worker --loglevel INFO -E -n embedding -c 1 -Q embedding
    sleep 20
done
```

solderzzc, 2019-03-26:
CLUSTER_REDIS_ADDRESS also need to be set in `detector`, please try the latest code which has `env/common.sh` for environment setup in one place.

solderzzc, 2019-03-26:
@yufeng2015 you can use search to check where this environment ver need to be set:
https://github.com/SharpAI/DeepCamera/search?q=CLUSTER_REDIS_ADDRESS&unscoped_q=CLUSTER_REDIS_ADDRESS

solderzzc, 2019-03-26:
Close for now, reopen if needed.

closed 2019-03-26 by solderzzc