SharpAI/DeepCamera #34: run source code on rk3288 failed

lambda-zhang, 2019-03-27:
followed https://github.com/SharpAI/DeepCamera/blob/master/docs/RunOnRK3288.md

```
rm: cannot remove '/root/.chushlogin': No such file or directory
rm: cannot remove '/root/.chushlogin'/bin/bash: line 0: cd: /data/data/com.termux/files/home/sharpai/src/embedding: No such file or directory
/bin/bash: line 0: cd: /data/data/com.termux/files/home/sharpai/src/face_detection: No such file or directory
: No such file or directory
WARNING: linker: libatomic.so: unused DT entry: type 0x6ffffef5 arg 0x398
WARNING: linker: /bin/bash: line 0: cd: /data/data/com.termux/files/home/sharpai/src/embedding: No such file or directory
```

```
[2019-03-27 15:39:58,898: WARNING/ForkPoolWorker-1] error of loading net2
[2019-03-27 15:39:58,905: ERROR/ForkPoolWorker-1] Signal handler <function setup at 0xadd397f0> raised: ImportError('No module named mxnet',)
Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/lib/python2.7/site-packages/celery/utils/dispatch/signal.py", line 227, in send
    response = receiver(signal=self, sender=sender, **named)
  File "upload_api-v2.py", line 1651, in setup
    mod = FaceProcessing.init_embedding_processor()
  File "/data/data/com.termux/files/home/DeepCamera/src/embedding/FaceProcessing.py", line 153, in init_embedding_processor
    import mxnet as mx
ImportError: No module named mxnet

```

solderzzc, 2019-03-27:
Please check the script to run src on aarch64. didnt ported to rk3288. we
dont need to support non opencl env on rk3288. just set HAS_OPENCL to true
will skip the mxnet importing.

followed

lambda-zhang, 2019-03-27:
https://github.com/SharpAI/DeepCamera/pull/35

closed 2019-03-27 by solderzzc