SharpAI/DeepCamera #55: Support for Jetson nano?

SaddamBInSyed, 2020-09-27:
Hi there.

thanks for your works.

I am a newbie to this repo. I would like to know whether I can run this DeepCamera in my Jetson nano?

please advise.

solderzzc, 2020-09-28:
hi, @SaddamBInSyed welcome to community, DeepCamera can run in Jetson Nano by source code but need change scripts to fit system. 
https://github.com/SharpAI/DeepCamera/blob/master/scripts/start_service.sh
I can built ARM64v8 docker to save the manually effort in weeks. :)

SaddamBInSyed, 2020-09-28:
thanks for your response.
I will wait for the same.

solderzzc, 2020-09-28:
just finished the build file on branch `arm64v8`, Travis CI will build images, then you can have a try 

https://travis-ci.org/github/SharpAI/DeepCamera

SaddamBInSyed, 2020-09-28:
@solderzzc 
thanks for sharing and letting me know.

solderzzc, 2020-10-29:
@SaddamBInSyed did you have chance to run DeepCamera on Nano? you can use DeepCamera/run-on-arm64v8.sh to start services, let me know if any issues.

solderzzc, 2020-12-07:
close for now

closed 2020-12-07 by solderzzc