SharpAI/DeepCamera #65: Install checkup for requirements - Suggestion/Improvement

staberas, 2021-11-07:
Enviroment : Jetson nano - 2GB version

The ./run-on-nano.sh script  requires a check if the user has :

- Has installed the correct version of  docker-compose (1.27.4) , i had the 1.17 version and had to manually find from docker website the correct version, i also had problem with pip so a check up if docker is correctly installed before it executes the commands would be nice.
-  After running ./run-on-nano.sh start , jetson stopped responding since its 29GB memory got filled up,  a check for sufficient storage before installing would be nice 

Thanks for the great job you did i hope you implement my suggestions/improvements

zhichao2021, 2021-11-07:
Thanks for the advise, I'll add the env checks.

For issue 2, do you have 4G memory version of Nano for testing, I haven't test on 2GB memory version of Nano yet.
I think the 29GB storage is big enough.

solderzzc, 2021-11-07:
Thanks for the advise, I'll add the env checks.

For issue 2, do you have 4G memory version of Nano for testing, I haven't test on 2GB memory version of Nano yet.
I think the 29GB storage is big enough.

staberas, 2021-11-07:
How much hard disk space does docker requires? (my jetson disk space is actually filled with other vision projects)  i have the 2gb ram version but expanded with swap memory

solderzzc, 2021-11-08:
@staberas  I'll checkup the benchmark of 2gb Nano since I have one but haven't run deepcamera on it. If it fails, I'll try to optimize it later. 
For disk space, it should be something like less than 4-6GB.

staberas, 2021-11-09:
Managed to installed and it just about works, its freezes and occupies 80-90% of memory which triggers swapping, this means  it kills the vnc viewer and you can only control it via ssh. Other than that shinobi is accesible on the network though a bit slow to load. 

For my application i just need the serverAPI ( ./start-cloud.sh ) for face recognition , since i have a sipeed M1N to do the face detection and cropping for the serverAPI to learn and recognise. Im i correct ?

solderzzc, 2021-11-09:
API server provides label interface and management GUI for devices, face recognition doesn't run on API Server.
Face recognition is running on device, it opens port 3000 to accept image to be processed. 

For your reference:
https://github.com/SharpAI/DeepCamera/blob/nano/src/detector/index.js#L917

solderzzc, 2021-11-09:
Following is the Android Java code to upload face image and build json message then send to DeepCamera for face recognition:
https://github.com/SharpAI/AI_Camera/blob/master/sharpaiLib/src/main/java/org/sharpai/lib/Detection.java#L657

solderzzc, 2022-08-18:
Release v3.4.3 addressed installation checkup:
https://github.com/SharpAI/DeepCamera/releases/tag/v3.4.3

solderzzc, 2022-08-25:
Fixed it with v3.4.3 release. @staberas thanks for your advise, we released a cli command line which can be easily installed by pip install, then one command can start the service instead of start with 'git clone'.
```
sharpai-cli deepcamera start
```

closed 2022-08-25 by solderzzc