SharpAI/DeepCamera #67: local variable 'person_count' referenced before assignment

roykroy10, 2022-01-20:
variable "person_count" referenced before assignment.
launched current nano branch on jetson nano
added token for telegram bot.
ran: run-on-nano.sh

face_detector      |   File "/root/detection/detector.py", line 220, in detect
face_detector      |     if person_count == 0:
face_detector      | UnboundLocalError: local variable 'person_count' referenced before assignment
face_detector      | [2022-01-16 21:53:49,867: INFO/MainProcess] Received task: upload_api-v2.detect[02779ca3-1f3d-4e21-968b-eda66e6aa7cf]   expires:[2022-01-16 21:54:19.000863+00:00]
detector_plugin    | error
detector_plugin    | null
detector_plugin    | add detect_task failed!!
face_detector      | [2022-01-16 21:53:49,920: ERROR/ForkPoolWorker-1] Task upload_api-v2.detect[02779ca3-1f3d-4e21-968b-eda66e6aa7cf] raised unexpected: UnboundLocalError("local variable 'person_count' referenced
before assignment",)
face_detector      | Traceback (most recent call last):
face_detector      |   File "/usr/local/lib/python2.7/dist-packages/celery/app/trace.py", line 374, in trace_task
face_detector      |     R = retval = fun(*args, **kwargs)
face_detector      |   File "/usr/local/lib/python2.7/dist-packages/celery/app/trace.py", line 629, in __protected_call__
face_detector      |     return self.run(*args, **kwargs)
face_detector      |   File "worker.py", line 52, in detect
face_detector      |     result = m.detect(image_path,trackerid, ts, cameraId, face_filter=None)
face_detector      |   File "/root/detection/detector.py", line 220, in detect
face_detector      |     if person_count == 0:
face_detector      | UnboundLocalError: local variable 'person_count' referenced before assignment

solderzzc, 2022-01-20:
@roykroy10 thank for reporting this issue, let me follow up.
It seems you already connect to a local RTSP camera stream through Shinobi GUI, right?

solderzzc, 2022-01-20:
@roykroy10 
I am wondering why this bug is still there, but I submit a quick fix, do you have chance to run code on branch **person_count_fix** and let me know the test result, if the bug is fixed by this PR, I'll merge.

roykroy10, 2022-01-21:
Hey @solderzzc ,
Thanks for the quick reply.
Yes, that's correct, I was using a RTSP stream from the local network. configured from shinobi GUI.
I appreciate the quick fix I'll test it and update soon.

Thanks
Roy

roykroy10, 2022-01-25:
Hey, it works now thanks.
Is it possible to pass a chat_id that's allowed to receive messages?

solderzzc, 2022-01-25:
@roykroy10 nice to know it works. Since you already configured the telegram token, after you boot up your deep camera, you can send a message to your telegram bot, then deep camera will save the chat id which deep camera could send message to. Currently, I'm not sure if we need to configure the chat id manually..

closed 2022-03-29 by solderzzc