SharpAI/DeepCamera #70: Face images are grey on the browser interface. Looks like API server is presenting image without jpg extension.

thartman69, 2022-03-20:
Great project!  Well done.

I am running the master image on a Ubuntu 18.04 VM. (virtualised with Proxmox PVE)
Using a Foscam 720p camera that connects fine via RTSP and I can see a picture.
Flower shows that the images are being processed and uploaded to the Minio bucket.
Connecting to API server port 3000 I see the activity but only grey boxes where the face pictures should be.

If I download the picture from Minio and rename it to *,jpg I can view the picture and it is a face.

In Flower log I can see ['/opt/nvr/detector/images/deepeye_1647774820586.jpg', 1647774820587, 1647774820587, 'pBrIitBTIr']

[image]

For some reason the API server is not presenting the file with the JPG extension.  Is this a *known* Minio bug, or other?

thartman69, 2022-03-22:
Hi SImba, thanks for responding.  I'm using Chrome on a PC and have also tested with other browsers on Mac, PC and IOS.  I am using the master docker image locally.  As far as I can tell I've configured everything correctly.  Face images are popping in the bucket.  If I rename a file to jpg or png they are displayed in the browser, so it looks like FFMPEG extracts and encodes the base64 image correctly from my old 720p Foscam for upload into Minio.  I've been digging into API Server code to see if it conforms to security settings.  There was a suggestion that Chrome had tightened security around files without extensions being presented.  Or forcing it in htaccess... I have not given up. :)  I will try the public testing server today.

solderzzc, 2022-03-22:
Hi, @thartman69 
Thanks for digging into this problem, I believe htaccess suggestion was introduced by browser upgrade. Let me also modify code when upload face image to minio to include the extension .jpg, I believe public server also has the same issue since it's the same docker image.

thartman69, 2022-03-22:
That's be awesome.  Can't wait to see it all in action.

solderzzc, 2022-03-29:
Hi, @thartman69 
I'm not able to reduplicate this issue in my setup, my browser is safari version 15.3, I'm able to load image.

I modified the code to upload image with png extension name. 
If something still wrong, please also send me the browser inspection log.

thartman69, 2022-03-29:
awesome, I'll rebuild it tonight and let you know.

thartman69, 2022-03-30:
Ah, I see you modified the Nano code, I'm using the Master branch because I'm using it in a VM.  Should I switch to the Nano branch?

solderzzc, 2022-03-30:
Yep, you're right, please switch to nano branch.

thartman69, 2022-03-30:
Excellent, that's even better because I want to use the Telegram feature.  Thank you, Simba!

solderzzc, 2022-03-30:
Great, let me know if there’s any issue.

thartman69, 2022-03-30:
The nano release won't install nicely on a KVM VM, I get errors in Flower showing up and only one thread.  I'll see if I can fork the project and apply the nano changes you made to the master release to get it working,

closed 2022-08-25 by solderzzc