SharpAI/DeepCamera #71: NameError: global name 'm' is not defined

WK-ING, 2022-04-26:
launched current nano branch on jetson nano.
added token for telegram bot.
connected to a RTSP camera stream in local network through Shinobi GUI.
ran: run-on-nano.sh

face_detector      | [2022-04-26 11:25:47,895: ERROR/ForkPoolWorker-1] Task upload_api-v2.detect[5ac8a3ac-0fa1-4bcf-99dd-cf074bc2abb0] raised unexpected: NameError("global name 'm' is not defined",)
face_detector      | Traceback (most recent call last):
face_detector      |   File "/usr/local/lib/python2.7/dist-packages/celery/app/trace.py", line 374, in trace_task
face_detector      |     R = retval = fun(*args, **kwargs)
face_detector      |   File "/usr/local/lib/python2.7/dist-packages/celery/app/trace.py", line 629, in __protected_call__
face_detector      |     return self.run(*args, **kwargs)
face_detector      |   File "worker.py", line 52, in detect
face_detector      |     result = m.detect(image_path,trackerid, ts, cameraId, face_filter=None)
face_detector      | NameError: global name 'm' is not defined
face_detector      | [2022-04-26 11:25:48,063: INFO/MainProcess] Received task: upload_api-v2.detect[6ada6e39-f2db-4e38-832e-cbbdf38449a5]   expires:[2022-04-26 11:26:18.000059+00:00]
detector_plugin    | error
detector_plugin    | null
detector_plugin    | add detect_task failed!!

solderzzc, 2022-04-26:
It seems you are not running the latest code or containers, if you upgraded the code by switch branch or git pull, you can try delete docker containers and images then rerun the script. Let me know if it's your case.

WK-ING, 2022-04-27:
Thanks for your quick reply. Actually I didn't upgraded the code except for adding a telegram token. 
My environment:
```
nvidia-jetpack Version: Jetpack 4.6 [L4T 32.6.1]
CUDA Version: 10.2.300
OpenCV Version: 4.1.1 compiled CUDA: NO
TensorRT Version: 8.0.1.6
cuDNN Version: 8.2.1.32
```
I tried delete images and rerun the script, the problem is still here. I double-checked the log information again and found that there is an ImportError(libnvinfer.so.7) before that error. 
```
face_detector      | [2022-04-27 00:41:17,445: ERROR/ForkPoolWorker-1] Signal handler <function setup at 0x7fa1d4fe50> raised: ImportError('libnvinfer.so.7: cannot open shared object file: No such file or directory',)
face_detector      | Traceback (most recent call last):
face_detector      |   File "/usr/local/lib/python2.7/dist-packages/celery/utils/dispatch/signal.py", line 227, in send
face_detector      |     response = receiver(signal=self, sender=sender, **named)
face_detector      |   File "worker.py", line 34, in setup
face_detector      |     import detector as m
face_detector      |   File "/root/detection/detector.py", line 11, in <module>
face_detector      |     from mtcnn import TrtMtcnn
face_detector      |   File "/root/detection/mtcnn.py", line 6, in <module>
face_detector      |     import pytrt
face_detector      | **ImportError: libnvinfer.so.7: cannot open shared object file: No such file or directory**
```

This may explain why 'm' is none in the subsequent steps. 
libnvinfer.so.7 is also not be found in embedding docker.

```
embedding          | [2022-04-27 00:41:15,332: INFO/MainProcess] Connected to redis://guest@redis:6379/0
embedding          | [2022-04-27 00:41:15,357: INFO/MainProcess] mingle: searching for neighbors
embedding          | Traceback (most recent call last):
embedding          |   File "upload_api-v2.py", line 46, in <module>
embedding          |     import FaceProcessing
embedding          |   File "/data/runtime/src/FaceProcessing.py", line 15, in <module>
embedding          |     import torch
embedding          |   File "/usr/local/lib/python2.7/dist-packages/torch/__init__.py", line 81, in <module>
embedding          |     from torch._C import *
embedding          | **ImportError: libnvinfer.so.7: cannot open shared object file: No such file or directory**
```

To avoid that I miss anything related (like other warnings), I paste the full log information as follows:

```
starting...
48**********
Your Device UUID is echo 48**********
WARNING: The ADMIN_PASSWORD variable is not set. Defaulting to a blank string.
WARNING: The TIMEZONE_OFFSET variable is not set. Defaulting to a blank string.
Starting redis  ... done
Starting camera          ... done
Starting minio           ... done
Starting broker          ... done
Starting docker_flower_1 ... done
Starting face_detector   ... done
Starting embedding       ... done
Starting detector_plugin ... done
Attaching to redis, minio, camera, broker, docker_flower_1, face_detector, embedding, detector_plugin
flower_1           | [I 220427 00:41:11 command:139] Visit me at http://localhost:5555
flower_1           | [I 220427 00:41:11 command:144] Broker: redis://redis:6379/0
flower_1           | [I 220427 00:41:11 command:147] Registered tasks: 
flower_1           |     [u'celery.accumulate',
flower_1           |      u'celery.backend_cleanup',
flower_1           |      u'celery.chain',
flower_1           |      u'celery.chord',
flower_1           |      u'celery.chord_unlock',
flower_1           |      u'celery.chunks',
flower_1           |      u'celery.group',
flower_1           |      u'celery.map',
flower_1           |      u'celery.starmap']
flower_1           | [I 220427 00:41:11 mixins:229] Connected to redis://redis:6379/0
[... 355 lines cut ...]
detector_plugin    |   run `npm audit fix` to fix them, or `npm audit` for details
detector_plugin    | Wed, 27 Apr 2022 00:41:35 GMT node-telegram-bot-api deprecated Automatic enabling of cancellation of promises is deprecated.
detector_plugin    | In the future, you will have to enable it yourself.
detector_plugin    | See https://github.com/yagop/node-telegram-bot-api/issues/319. at module.js:653:30
detector_plugin    | init nedb global
detector_plugin    | init nedb global for gif
detector_plugin    | init nedb global for timeline
detector_plugin    | upload listerer is ready
detector_plugin    | wait detect queue init 
detector_plugin    | telegram token is 5258423136:AAEdDGNXCa7ESNDPbhcW4vcICPap1OuTZo8
detector_plugin    | Listening on port  3000
detector_plugin    | select database 15 to avoid conflict
detector_plugin    | The connection to celery cluster is connected
detector_plugin    | The connection to celery broker is connected
detector_plugin    | Bucket exists.
camera             | 2022-04-27T00:41:36+00:00 pluginInitiatorSuccess { f: 'init',
camera             |   plug: 'WorkAI',
camera             |   notice: 'Looks like you have the Motion plugin running. Don\'t forget to enable <b>Send Frames</b> to start pushing frames to be read.',
camera             |   type: 'detector',
camera             |   pluginKey: 'asdfasdfasdf12341234ad79fgsx' } 
camera             | 2022-04-27T00:41:36+00:00 Connected to plugin : Detector - WorkAI - detector  
detector_plugin    | API Key from server is WorkAI
detector_plugin    | init_monitor
detector_plugin    | init_monitor
```

WK-ING, 2022-04-27:
I searched for some solutions. Some says that it's because TensorRT in my system is not compatible with the app. Do you have any restrictions about TensorRT Version in this app?

solderzzc, 2022-04-27:
Oh, it seems you need install jetpack 4.5.1 which has tensorrt 7

WK-ING, 2022-04-27:
> Oh, it seems you need install jetpack 4.5.1 which has tensorrt 7
> […](#)

Oh no. I think that's the problem... And unfortunately there isn't a safe and direct way to [downgrade the Jetpack](https://forums.developer.nvidia.com/t/downgrade-jetpack/169373) I think :'( ? 
Many thanks for your always quick reply. Have a good day!

closed 2022-04-27 by WK-ING