SharpAI/DeepCamera #72: Cannot see given setting on the 8080 port over shinobi page on web browser

KenaHemnani, 2022-06-07:
I cannot see the setting options to give camera ip and user password on shinobi webpage.
https://drive.google.com/drive/folders/1WimiJtyC_GNgdqPHFhKISS5G8M14NEXE?usp=sharing

Can you tell me what is the issue?

solderzzc, 2022-06-07:
You can setup username and password by visiting connection section:
[image]

KenaHemnani, 2022-06-08:
Hi @solderzzc , Thank you for the reply.
But I am not able to see these settings inside connection section, I am only getting the options given in images here:
https://drive.google.com/drive/folders/1WimiJtyC_GNgdqPHFhKISS5G8M14NEXE?usp=sharing

KenaHemnani, 2022-06-08:
This is what I am able to see inside connection section:
[image]

Please, let me know. What is the problem? I would be a great help!

solderzzc, 2022-06-08:
You can set Automatic to ‘No’ to setup username and password manually.

KenaHemnani, 2022-06-08:
Hi @solderzzc , Thank you! Now I am able to see the option. After giving host, user and password. I am able to see the black screen as below. Also, I have deleted the moniter once and added new then. Will it makes difference? 

[image]

Can you tell how to troubleshoot this? 
I am currently using my android mobile as Web-cam.

solderzzc, 2022-06-21:
@KenaHemnani 
when you have this page, make sure you have a url can be played by ffplay, then you can copy paste the url into Full URL Path:

[image]
/

KenaHemnani, 2022-06-22:
@solderzzc , Thank you for help! I am able to get feed in shinobi over http://localhost:8080/ . 
*Are we required to connect our camera with shinobi running over http://165.232.62.29:8080/ ? 
*How to see those images on GUI running on server ?

solderzzc, 2022-06-22:
@KenaHemnani 
* You don't have to connect camera with shinobi running over http://165.232.62.29:8080/
* Please follow the [readme instruction](https://github.com/SharpAI/DeepCamera#label-on-web-gui-train-face-recognition-model-on-device), you need signup with http://165.232.62.29:3000, then add your device serial-id to cloud server. then you will be able to see the images.

KenaHemnani, 2022-06-23:
Hi @solderzzc , Thank you very much for responding. 
On the localhost:8080 when accessing shinobi, I can record videos and take snapshots. However, images cannot be previewed but I can see preview of videos and videos are saved inside DeepCamera/docker/videos folder. But, I cannot see the video/images on server http://165.232.62.29:3000/ (The device serial id is added to cloud server.)
Can you tell what might be reason for this? How I can troubleshoot this?

Are we required to take snapshots using shinobi or record videos? However, Snapshots are not able to preview over shinobi nor saved on local on any folder.

solderzzc, 2022-06-23:
@KenaHemnani we don't have to worry about shinobi after you have seen preview pic on it, we use it for camera configuration. you should follow the [instruction](https://github.com/SharpAI/DeepCamera#label-on-web-gui-train-face-recognition-model-on-device) to label face on cloud server:

KenaHemnani, 2022-06-23:
@solderzzc , but I am not able to see pic on GUI running over cloud server i.e, on http://165.232.62.29:3000/ . What can be reason for this ? How can I troubleshoot this ?

solderzzc, 2022-06-23:
@KenaHemnani Please provide the system log and your hardware configuration, are you using jetson nano?

KenaHemnani, 2022-06-24:
Currently I am running DeepCamera on linux machine(with 16GB Ram) using command `sudo ./run-on-linux.sh start`. And using android mobile camera as webcam(IP camera). Then starting cloud using `sudo ./start-cloud.sh start`. 
**Systems logs while running `sudo ./run-on-linux.sh start`:**
+ IFNAME=eth0
+ YML=./docker/docker-compose-x86.yml
+ '[' startx == x ']'
+ case $1 in
+ echo starting...
starting...
+ gen_ro_serial
+ UUIDFILE=./docker/workaipython/ro_serialno
+ MAC=
+ '[' -e ./docker/workaipython/ro_serialno ']'
++ cat ./docker/workaipython/ro_serialno
+ MAC=rd2910919898
+ '[' rd2910919898x == x ']'
+ '[' 12 '!=' 12 ']'
+ echo rd2910919898
rd2910919898
+ '[' xrd2910919898 = x ']'
+ docker-compose -f ./docker/docker-compose-x86.yml stop
Stopping broker          ... done
Stopping api_server      ... done
Stopping minio           ... done
Stopping detector_plugin ... done
Stopping embedding       ... done
Stopping docker_flower_1 ... done
Stopping face_detector   ... done
Stopping redis           ... done
Stopping camera          ... done
+ docker-compose -f ./docker/docker-compose-x86.yml up
WARNING: The ADMIN_PASSWORD variable is not set. Defaulting to a blank string.
WARNING: The TIMEZONE_OFFSET variable is not set. Defaulting to a blank string.
Starting camera       ... done
Recreating minio         ... done
Recreating broker        ... done
Recreating api_server    ... done
Starting redis        ... done
Starting face_detector     ... done
Starting docker_flower_1 ... done
Recreating embedding       ... done
Recreating detector_plugin ... done
Attaching to camera, redis, minio, broker, api_server, docker_flower_1, face_detector, embedding, detector_plugin
detector_plugin    | export API_SERVER_ADDRESS='165.232.62.29'
detector_plugin    | export API_SERVER_PORT='3000'
detector_plugin    | export AWS_ACCESS_KEY='QUA2IU17RHOKE6NUZ7T2'
detector_plugin    | export AWS_BUCKET='faces'
detector_plugin    | export AWS_END_POINT='165.232.62.29'
detector_plugin    | export AWS_PORT='9000'
detector_plugin    | export AWS_READABLE_PREFIX='http://165.232.62.29:9000/faces/'
detector_plugin    | export AWS_SECRET_KEY='MQniMc5K3lsRbv9OPaUbJN9ft2eTQJ1rh4Yx6C17'
detector_plugin    | export AWS_USE_SSL='false'
detector_plugin    | export BIGGEST_FACE_ONLY_MODE='0'
detector_plugin    | export CLUSTER_CONCURRENCY='80'
detector_plugin    | export CLUSTER_REDIS_ADDRESS='redis'
detector_plugin    | export CLUSTER_REDIS_PORT='6379'
detector_plugin    | export DEEP_ANALYSIS_MODE='0'
detector_plugin    | export ENABLE_STATIC_OBJECT_FILTER='false'
detector_plugin    | export GIF_UPLOADING='1'
detector_plugin    | export HOME='/root'
detector_plugin    | export HOSTNAME='e29412c9b937'
detector_plugin    | export IMAGE_DIR='/opt/nvr/detector/images'
detector_plugin    | export MINIMAL_FACE_RESOLUTION='60'
detector_plugin    | export MINIO_BROWSER='off'
detector_plugin    | export MQTT_BROKER_ADDRESS='165.232.62.29'
detector_plugin    | export MQTT_BROKER_PORT='1883'
detector_plugin    | export MQTT_BROKER_TRANSPORT='tcp'
detector_plugin    | export ONE_KNOWN_PERSON_BYPASS_QUEUE_MODE='1'
detector_plugin    | export PATH='/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin'
detector_plugin    | export PWD='/opt/nvr'
detector_plugin    | export REALTIME_STRANGER_SDK_MESSAGE='1'
[... 788 lines cut ...]
api_server    | I20220624-04:59:14.797(0)?     retryWait: 60000,
api_server    | I20220624-04:59:14.797(0)?     retried: 0,
api_server    | I20220624-04:59:14.798(0)?     retryBackoff: 'constant',
api_server    | I20220624-04:59:14.798(0)?     retryUntil: Sat Sep 13 275760 00:00:00 GMT+0000 (UTC),
api_server    | I20220624-04:59:14.798(0)?     repeats: 0,
api_server    | I20220624-04:59:14.798(0)?     repeatWait: 300000,
api_server    | I20220624-04:59:14.799(0)?     repeated: 0,
api_server    | I20220624-04:59:14.799(0)?     repeatUntil: Sat Sep 13 275760 00:00:00 GMT+0000 (UTC),
api_server    | I20220624-04:59:14.799(0)?     after: Fri Jun 24 2022 05:04:14 GMT+0000 (UTC),
api_server    | I20220624-04:59:14.799(0)?     progress: { completed: 0, total: 1, percent: 0 },
api_server    | I20220624-04:59:14.800(0)?     depends: [],
api_server    | I20220624-04:59:14.800(0)?     resolved: [],
api_server    | I20220624-04:59:14.800(0)?     log: [ [Object], [Object] ] } ]
api_server    | I20220624-04:59:14.944(0)? []
api_server    | I20220624-04:59:18.331(0)? ----setup smtp server---
api_server    | I20220624-04:59:18.340(0) (percolate_synced-cron.js:87) SyncedCron: Scheduled "send report email 12:00 am every day" next run @Fri Jun 24 2022 04:59:18 GMT+0000 (UTC)
api_server    | => Started your app.
api_server    | 
api_server    | => App running at: http://localhost:3000/
api_server    | I20220624-05:00:00.008(0) (percolate_synced-cron.js:87) SyncedCron: Starting "send report email 12:00 am every day".
api_server    | I20220624-05:00:00.011(0)? send email out
api_server    | I20220624-05:00:00.012(0) (percolate_synced-cron.js:87) SyncedCron: Finished "send report email 12:00 am every day".
api_server    | I20220624-05:04:29.353(0)? send offline notification
api_server    | I20220624-05:04:29.354(0)? { clientId: 'test_client_id_1111' }
api_server    | I20220624-05:04:29.355(0)? send offline notification to test_client_id_1111

solderzzc, 2022-06-24:
@KenaHemnani  For running on X86 Linux, you need checkout master branch, please make another clean folder to have master branch, then run the command again. When you use X86 version, after you configure camera connection, you can go to localhost:3000 for GUI.

KenaHemnani, 2022-06-27:
@solderzzc Thank you for clarifying this. I have made a new folder and clone the repo and switched to master branch. 
**Following are System logs while running** `sudo ./run-on-linux.sh start`:
[image]

**Following are Systems logs while running** `sudo ./start-cloud.sh start`:
[image]

Going to localhost:3000 using http://localhost:3000/
Then this url is not working:
[image]

KenaHemnani, 2022-06-27:
Also, I have tried doing API_SERVER_ADDRESS=127.0.0.1 in servers.env. But the api could not run. Also, let me know what should be variables values in aws.env and servers.env files.

solderzzc, 2022-06-27:
@KenaHemnani you only need to run **sudo run-on-linux.sh start**, it takes several minutes to install dependencies as you already observed. You don't need to run **start-cloud.sh** if you have already launched with **run-on-linux.sh**. You don't have to modify servers.env. To load image on gui, you need configure AWS_READABLE_PREFIX=http://minio:9000/faces/ in **aws.env**, you only need to change AWS_READABLE_PREFIX=http://<IP_ADDR>:9000/faces/ IP_ADDR is an address you can access when you are using browser to open server running at.

KenaHemnani, 2022-07-15:
Hi @solderzzc, Thank you for the response. After the above suggested changes I am able to see detection result via. message on GUI at http://165.232.62.29:3000/message sayin "AI saw kena", since it was previously labelled on cloud when minio was running at <IP_ADDR>=165.232.62.29 i.e, AWS_READABLE_PREFIX. But, I am not able to see images over "Name Unknown" tab at GUI first page(It shows gray images). Also I am not able to label them. 
How to label and train model on linux system?

solderzzc, 2022-07-15:
@KenaHemnani can you share your configuration files with modification?

solderzzc, 2022-07-18:
@KenaHemnani It's caused by server address mismatching, the IP address of AWS_END_POINT and AWS_READABLE_PREFIX must be the same. 
```
AWS_END_POINT=165.232.62.29
AWS_READABLE_PREFIX=http://192.168.0.71:9000/faces/
```

KenaHemnani, 2022-07-19:
@solderzzc ,Thank your for the response. After changing aws.env as following:

AWS_END_POINT=192.168.0.71
AWS_PORT=9000
AWS_USE_SSL=false
AWS_ACCESS_KEY=QUA2IU17RHOKE6NUZ7T2
AWS_SECRET_KEY=MQniMc5K3lsRbv9OPaUbJN9ft2eTQJ1rh4Yx6C17
AWS_BUCKET=faces
AWS_READABLE_PREFIX=http://192.168.0.71:9000/faces/ 

I am able to see the captured images at minio http://192.168.0.71:9000/faces/  . But, in the GUI the images are still showing up gray over 'Name Unknown' and 'Timeline' though it is detecting faces and giving message 'AI saw <person name>'. These detection are because the model was trained previously. How can we label and train the facial images at the local system i.e, in this case we are using linux machine.

solderzzc, 2022-07-20:
@KenaHemnani You should be able to get a clean environment by deleting all docker and local files, then you can use the clean local deployment from the beginning. Make sure you have created **faces** bucket with your minio server. The model will be trained again when you label them with the local deployment again. If it still not works, you can attach browser inspector log here. 
Whenever you label an image, a transfer learning script will be called and your local model will be updated.

KenaHemnani, 2022-07-22:
Hi @solderzzc , Thank you for the response. 
I have deleted docker and intsalled again and run everything from start.
Case-I:
For below configuration files:
aws.env:
AWS_PORT=9000
AWS_READABLE_PREFIX=http://192.168.0.71:9000/faces/

servers.env:
MQTT_BROKER_ADDRESS=192.168.0.71
MQTT_BROKER_PORT=1883
MQTT_BROKER_TRANSPORT=tcp
API_SERVER_ADDRESS=192.168.0.71
API_SERVER_PORT=3000

Then the terminal is saying app is running at localhost:8000:
[image]
Browser inspect: (at  http://localhost:3000)
[image]

Case-II:
When running only ./run-on-linux.sh with above configuration of aws.env:
Browser inspect at (at http://165.232.62.29:3000/)
[image]
Continuing...
[image]

**Here, we can see images but they are gray and cannot label it.

Also, how to run the GUI at localhost:3000 ???**

solderzzc, 2022-07-27:
Issue 1:
Your local browser should be point to http://192.168.0.71:3000 to get your local label interface work.

Issue 2:
If your api server is set to cloud server, you should also change aws.env to upload image to the cloud server to get rid of cross domain issue.

KenaHemnani, 2022-07-28:
Hi @solderzzc , Thank you for the response.
In continuation to issue-1:
I am Still getting the error. Please let me know what else is required.

[image]

solderzzc, 2022-07-28:
Private version is not supported for open-source version. please use cloud server.

solderzzc, 2022-08-25:
The installation pain point was solved by the latest release, now you can use command line to install application:
```
sharpai-cli deepcamera start
```

solderzzc, 2022-09-01:
Hi @KenaHemnani 
SharpAI team decided to bring local deployment with `sharpai-cli` based on your feedback:

https://github.com/SharpAI/DeepCamera/blob/master/README.md#local-deployment-is-ready-for-testing-now

closed 2022-11-29 by solderzzc