SharpAI/DeepCamera #96: Support HA OS on ODROID N2+ (aarch64 support)

apicht, 2022-10-21:
**Is your feature request related to a problem? Please describe.**
Unable to run on this platform: `Your platform is not supported`

**Describe the solution you'd like**
- Support for the aarch64 platform.
- Probably needs a separate request, but it would also be nice to have this packaged as a Home Assistant Add-on.

**Describe alternatives you've considered**
Only current alternative is to run on something else.

**Additional context**
Current kernel info:
```
# uname -a
Linux bbe2336a629c 5.15.72 #1 SMP PREEMPT Fri Oct 7 09:49:23 UTC 2022 aarch64 GNU/Linux
```

solderzzc, 2022-10-21:
Hi, @apicht 

I've submitted a bugfix, please upgrade and run command again:
```
pip3 install sharpai-hub --upgrade
sharpai-cli yolov7_reid start
```

Let me know if any issues.

I'll look into the API of HA for add-on, it's a good idea.

solderzzc, 2022-10-21:
Make sure sharpai-hub is upgraded to 0.1.42:
```
Installing collected packages: sharpai-hub
  Found existing installation: sharpai-hub 0.1.41
    Uninstalling sharpai-hub-0.1.41:
      Successfully uninstalled sharpai-hub-0.1.41
Successfully installed sharpai-hub-0.1.42
```

apicht, 2022-10-21:
Thanks, it now gets past that error. I have a new issue (filesystem-related) but it appears to be specific to my environment. Also, after glancing at the compose file, it appears yolov7_reid is going to run its own instance of homeassistant which probably isn't going to work in this environment. I may need to fork and tinker to get this running.

```
# sharpai-cli yolov7_reid start
The latest docker-compose.yml has downloaded
You haven't configured labelstudio service, SharpAI CLI will help you go through:                
- 1. Pulling latest docker images from docker-hub, time may vary depending on you network situation...
Pulling latest docker images from docker hub...
WARNING: The TIMEZONE_OFFSET variable is not set. Defaulting to a blank string.
Pulling redis          ... done
Pulling etcd           ... done
Pulling labelstudio    ... done
Pulling minio          ... done
Pulling milvus         ... done
Pulling detector       ... done
Pulling home-assistant ... done
- 2. Starting Labelstudio service..WARNING: The TIMEZONE_OFFSET variable is not set. Defaulting to a blank string.
Creating network "yolov7_reid_default" with the default driver
Creating labelstudio ... error
.
ERROR: for labelstudio  Cannot start service labelstudio: error while creating mount source path '/root/.sharpai/yolov7_reid/volumes/labelstudio-data': mkdir /root/.sharpai: read-only file system

ERROR: for labelstudio  Cannot start service labelstudio: error while creating mount source path '/root/.sharpai/yolov7_reid/volumes/labelstudio-data': mkdir /root/.sharpai: read-only file system
ERROR: Encountered errors while bringing up the project.
............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
```

solderzzc, 2022-10-21:
1. I use user account instead of `root`. 
2. It seems a setup guideline with sharpai-cli is not running correctly, in this case, you can `rm -rf ~/.sharpai` then re-run the cli command again
3. Home-assistant integration process is:
    1. Copy sharpai addon to home-assistant. https://github.com/SharpAI/DeepCamera/blob/master/src/home-assistant-py/Dockerfile#L2
    2. create a shared path with docker-compose so addon can send file path to detector directly instead of file upload https://github.com/SharpAI/DeepCamera/blob/master/src/home-assistant-py/sharpai/image_processing.py#L287