SharpAI/DeepCamera #98: mac not working

wang2366, 2023-04-18:
Your platform dose not support yet, please file a bug: arm
Your platform is not supported, please file an issue on github for feature request: https://github.com/SharpAI/DeepCamera/issues

solderzzc, 2023-05-12:
It seems you are using Apple M2/M1 chip.

anu1097, 2023-06-22:
How to circumvent this ? 
I tried checking out the code locally and tried running from the docker file in docker folder. 

shareai/shinobi:arm64v8_latest, shareai/embedding:arm64v8_latest and 
shareai/face_detector:arm64v8_latest didn't even start, no logs either ? 

tried running from individual source folder as well. 
For camera. Some file is missing in source code ./nvr_releases/2018.03.15.14.50.tar.gz

solderzzc, 2023-06-22:
@anu1097 
I'm not sure if Mac's M1/M2 share the same architecture as embedding system.

Can you try to build the docker images with the dockerfile created for Raspberry Pi which has ARM processor?

You need go to folder:
```
src/yolov7_reid
```
then:
```
docker compose build -f docker-compose-arm64.yml
docker compose up -f docker-compose-arm64.yml
```

anu1097, 2023-06-27:
So I managed to run this dockerfile via this command - 

`docker compose -f docker-compose-arm64.yml up`

One issue though, detector image although its running I see nothing on localhost:3000, or ports 8000, 9000 or 5901. 
I also tried an alternative way tried to run the detector file locally commented out the part from docker compose file. 

Now I see it waiting for milvus to run. However Milvus is running.

Here are the logs - 

```
2023-06-27 16:09:58 [2023/06/27 10:39:58.175 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:01 [2023/06/27 10:40:01.175 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:04 [2023/06/27 10:40:04.176 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:04 [2023/06/27 10:40:04.310 +00:00] [INFO] [datanode/data_node.go:817] [DataNode.GetCompactionState]
2023-06-27 16:10:04 [2023/06/27 10:40:04.310 +00:00] [DEBUG] [datanode/data_node.go:843] ["Compaction results"] [results="[]"]
2023-06-27 16:10:07 [2023/06/27 10:40:07.175 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:10 [2023/06/27 10:40:10.175 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:13 [2023/06/27 10:40:13.175 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:14 [2023/06/27 10:40:14.310 +00:00] [INFO] [datanode/data_node.go:817] [DataNode.GetCompactionState]
2023-06-27 16:10:14 [2023/06/27 10:40:14.310 +00:00] [DEBUG] [datanode/data_node.go:843] ["Compaction results"] [results="[]"]
2023-06-27 16:10:16 [2023/06/27 10:40:16.176 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:19 [2023/06/27 10:40:19.175 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:22 [2023/06/27 10:40:22.176 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
2023-06-27 16:10:24 [2023/06/27 10:40:24.307 +00:00] [INFO] [datanode/data_node.go:817] [DataNode.GetCompactionState]
2023-06-27 16:10:24 [2023/06/27 10:40:24.307 +00:00] [DEBUG] [datanode/data_node.go:843] ["Compaction results"] [results="[]"]
2023-06-27 16:10:25 [2023/06/27 10:40:25.176 +00:00] [INFO] [indexcoord/index_builder.go:163] ["index builder task schedule"] ["task num"=0]
```

I'm interested to check out how the flask application works. 

And I'm still trying to understand how all these various applications are being used. I got some idea, would appreciate some docs. 

I get that redis and etcd are assisting various services to easily share data, and various services are label-studio for uploading data/images to help in creating classifiers. Minio is for Object Storage and Milvus is for easily storing Vector images, HomeAssistant for managing Video Camera feed. But still fail to see where is detector's code being called from ?

Could use a little help in starting. Thanks

anu1097, 2023-06-27:
What is docker-novnc for ?

anu1097, 2023-06-27:
I finally was able to get localhost:3000 working, but I'm not sure of 1 thing, where to see the video after its processed. 
And how to use label-studio. It allows you pass any video url. So I was testing with 1.

solderzzc, 2023-06-29:
@anu1097 
docker-novnc is used for providing a desktop environment for the runtime, the yolov7 detector is running in docker, you can access the docker desktop with http://localhost:8000/
The video preview is provided by home-assistant, Home-Assistant is hosted at http://localhost:8123/
Labelstudio is hosted at http://localhost:8080/ for saving image and further model training.

anu1097, 2023-06-29:
So all those links are working, but how do I connect them. I have IP webcam working on one of my devices. 
I get that I can add rtsp link from that to homeassistant, but how do I use docker-novnc ?