SharpAI/DeepCamera #102: Support for AMD64 Family

kausarn3, 2023-08-29:
DeepCamera is not supported for AMD64 platform

kausarn3, 2023-08-29:
Please find my below observation :
PS C:\Users\xyz> sharpai-cli yolov7_reid start
Your platform dose not support yet, please file a bug: AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD
Your platform is not supported, please file an issue on github for feature request: https://github.com/SharpAI/DeepCamera/issues
PS C:\Users\xyz>

I got the above error while executing command (sharpai-cli yolov7_reid start)

my Pc configuration:
OS Name:                   Microsoft Windows 11 Pro
OS Version:                10.0.22621 N/A Build 22621
OS Manufacturer:           Microsoft Corporation
OS Configuration:          Standalone Workstation
OS Build Type:             Multiprocessor Free
System Manufacturer:       LENOVO
System Model:              82JW
System Type:               x64-based PC
Processor(s):              1 Processor(s) Installed.
                           [01]: AMD64 Family 25 Model 80 Stepping 0 AuthenticAMD ~1882 Mhz
BIOS Version:              LENOVO HHCN24WW, 24-11-2021
Total Physical Memory:     16,236 MB
Available Physical Memory: 8,567 MB
Virtual Memory: Max Size:  17,260 MB
Virtual Memory: Available: 6,614 MB
Virtual Memory: In Use:    10,646 MB

Thank you

Usher09896, 2023-08-29:
This issue should be fixed by new release:
https://github.com/SharpAI/sharpai_hub/releases/tag/0.1.43

please upgrade with following command:
```
pip3 install --upgrade sharpai-hub
```

Let's know if it fixes this issue.

kausarn3, 2023-08-31:
@Usher09896 after upgrading the script it is working now, thanks for your support.

closed 2023-08-31 by kausarn3