SharpAI/DeepCamera #103: jetson nano jetpack 4.6-b199 is not supported

han88, 2023-12-29:
Your platform dose not support yet, please file a bug: 
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Package: nvidia-jetpack
Version: 4.6-b199
Priority: standard
Section: metapackages
Maintainer: NVIDIA Corporation
Installed-Size: 199 kB
Depends: nvidia-cuda (= 4.6-b199), nvidia-opencv (= 4.6-b199), nvidia-cudnn8 (= 4.6-b199), nvidia-tensorrt (= 4.6-b199), nvidia-visionworks (= 4.6-b199), nvidia-container (= 4.6-b199), nvidia-vpi (= 4.6-b199), nvidia-l4t-jetson-multimedia-api (>> 32.6-0), nvidia-l4t-jetson-multimedia-api (<< 32.7-0)
Homepage: http://developer.nvidia.com/jetson
Download-Size: 29,4 kB
APT-Sources: https://repo.download.nvidia.com/jetson/t210 r32.6/main arm64 Packages
Description: NVIDIA Jetpack Meta Package

Is there a solution yet?
thanks in advance

han88, 2024-01-07:
Jetpack 4.6-b110 works...

solderzzc, 2024-01-08:
@han88  thanks for the feedback.

han88, 2024-01-09:
but the telegram bot doesn't function:

Exception in thread Thread-9:
Traceback (most recent call last):
File "/usr/lib/python3.6/threading.py", line916, in _bootstrap_inner
self.run()
File "/app/telegram_bot.py", line 49, in run
self.updater = Updater(self.token)
File "/usr/local/lib/python3.6/dist-packages/telegram/ext/updater.py", line_29
0, in __init__
else arbitrary_callback_data
File "/usr/local/lib/python3.6/dist-packages/telegram/ext/extbot.py", line 103
in __init__
private_key_password-private_key_password.
File "/usr/local/lib/python3.6/dist-packages/telegram/bot.py", line 191, in
init__
self.token = self._validate_token(token)
File "/usr/local/lib/python3.6/dist-packages/telegran/bot.py". line_355, in
alidate_token
raise InvalidToken()
telegram.error.InvalidToken: Invalid token
filename: /opt/nvr/detector/images/192_168_2_119_7dab08d0-47ff-4b6e-ae6a-134b25b
34f65.jpg, camera_id: 192_168_2_119
detect with frame

but the token I used is valid, works on other platforms!
Thanks

han88, 2024-01-09:
i missed to copy the id...

closed 2024-01-09 by han88