SharpAI/DeepCamera #107: Website link from github links to expired domain

RichardoC, 2025-02-01:
**Describe the bug**
Github has a link to [sharpai.github.io/DeepCamera/](https://sharpai.github.io/DeepCamera/) as the link for this project.
That web page redirects to AiCamera.io which has expired, and has a godaddy page offering it for sale.

Please update the link to the real project site.

**To Reproduce**
Github has a link to [sharpai.github.io/DeepCamera/](https://sharpai.github.io/DeepCamera/) as the link for this project.
visiting that web page redirects to AiCamera.io which has expired, and has a godaddy page offering it for sale.

**Expected behavior**
The project site listed on github should exist

**Screenshots**
N/A

**Desktop (please complete the following information):**
All

**Smartphone (please complete the following information):**
N/A

**Additional context**
Add any other context about the problem here.

solderzzc, 2025-02-01:
Thanks for posting the issue, let me check how to solve it.

solderzzc, 2025-02-02:
I've updated the website to use http://www.sharpai.org/DeepCamera/

RichardoC, 2025-02-03:
Thanks for sorting this

closed 2025-02-03 by RichardoC